es-ude/elastic-ai.creator #12: Shape Inference Warnings for ONNX exports

glencoe, 2022-01-05:
Currently we're producing the following warning
```
The shape inference of elasticai.creator::Wrapper type is missing, so it may result in wrong shape inference for the exported graph. Please consider adding it in symbolic function.
```
when exporting to onnx. I do not know atm how to add the shape inference to the symbolic function, but i think we should really do it.

miihwi, 2022-01-10:
There is this SetType function: https://pytorch.org/docs/stable/onnx.html. But it neither gets rid of the warning, nor can I see an easy way to leverage that

RodrigoWIssa, 2022-01-10:
There is a setType function: https://pytorch.org/docs/stable/onnx.html. This does not get rid of the warnings, nor can I think on a easy way to leverage that

glencoe, 2022-01-10:
> There is a setType function: https://pytorch.org/docs/stable/onnx.html. This does not get rid of the warnings, nor can I think on a easy way to leverage that

Damn... that's what i feared. Well, let's keep that open and see when/if someone finds a fix.

closed 2022-08-17 by julianhoever