es-ude/elastic-ai.creator #33: can the _prefix_use function be used for all libraries?

Annika10, 2022-01-21:
I'm just not sure about the prefix_use function because the lstm_cell.vhd has something like `LIBRARY work;` without use in before

_Originally posted by @Annika10 in https://github.com/es-ude/elastic-ai.creator/pull/31#discussion_r789653677_

glencoe, 2022-01-21:
> hmm.... i guess a library declaration would then look something like this:
> 
> ```
> library <lib>;
> use <lib>.<package>;
> use <lib>.<other_package>;
> ```
> 
> So i guess in the lstm file we're seeing two libraries, whereas we got only one here. With this in mind i don't think my proposed code makes sense.

just repeated from the linked comment

Annika10, 2022-01-26:
> hmm.... i guess a library declaration would then look something like this:
> ```
> library <lib>;
> use <lib>.<package>;
> use <lib>.<other_package>;
> ```
>
>
>
>
>
>
>
>
>
>
>
> So i guess [...]

Yes, you are right about this. But I talked with Chao about it and for now it is enough to use use the ieee library and not the work library. So I think we can stick to your proposed code for now

glencoe, 2022-01-26:
my problem with the proposed code is that its structure does not exactly match the vhdl specification.
From the specification:
```
library_clause ::= library logical_name_list ;
logical_name_list ::= logical_name { , logical_name }
logical_name ::= identifier
```

So what i called library header instead is actually the `library_clause`. In fact the proposed code should be

```python
class ContextClause:
  _library_clause = "library ieee;" # no idea if this is actually the library header, better look it up in the reference
  def __init__(self):
    self._use_clause = ["ieee.std_logic_1164.all", "ieee.numeric_std.all"]

  def __call__(self):
    yield self._library_clause
    yield from _add_semicolon_in_each_line(self.prefix_lines_with_USE(self.libs))
```

or even

```python
class ContextClause:
  def __init__(self, library_clause, use_clause):
    self._use_clause = use_clause
    self._library_clause = library_clause

  def __call__(self):
    yield from self._library_clause()  
    yield from self._library_clause()

class UseClause:
  def __init__(self, items: list[str]):
    self._items = items

  def __call__(self):
    def prefix_use(line: str):
      return f"use {line}"
    yield from _add_semicolon_in_each_line(map(prefix_use, items))

class LibraryClause:
  def __init__(self, logical_name_list: list[str]):
    self._logical_name_list = logical_name_list

  def __call__(self):
    yield "library {}".format(", ".join(self.logical_name_list))
```

glencoe, 2022-01-26:
We will have to reuse these code generators a lot, so it is crucial to make sure the terminology matches exactly the reference. For simplicity's sake we do not have to model every left side in the ebnfs as its own class and we do not have to model the full grammar, but only the subset we actually need. But it has to be correct in the formal sense, otherwise we'll run into trouble later.

Annika10, 2022-01-26:
okay I see your point and will change it like that. You have a different bock for the vhdl specification than me, it seem more detailed. The book I used do not mention the library definition that detailed :) I will add your book to the readme as well

Annika10, 2022-01-26:
So when I use your implementation approach and want to define the libraries, I do it like this?:

```
        lib = ContextClause(
            library_clause=LibraryClause(logical_name_list=["ieee", "work"]),
            use_clause=UseClause(
                selected_names=[
                    "ieee.std_logic_1164.all",
                    "ieee.numeric_std.all",
                    "work.all",
                ]
            ),
        )
```

glencoe, 2022-01-26:
yep, that doesn't mean the end user will have to do it like that though, but that's an issue for later

closed 2022-01-26 by Annika10