es-ude/elastic-ai.creator #52: Generating BRAMs to store weights and bias for lstm cell

SuperChange001, 2022-02-14:
Storing weight and bias on FPGA with BRAM is quite usual. And I want a feature from our `creator tool` that can generate a `rom_name.vhd` file by sending a floating array to it. Sure I also will specify the bit with for a fixed point and the bit with for its fractional part.

I made a [Jupyter-notebook](https://git.uni-due.de/embedded-systems/artificial-intelligence/hardwarecomponents/lstm/-/blob/main/python_scripts/example_of_generating_Memory.ipynb) and an [HDL template](https://git.uni-due.de/embedded-systems/artificial-intelligence/hardwarecomponents/lstm/-/blob/main/templates/rom_template.vhd). In the template, all arguments that can be modified are with a prefix `PARAM_`, e.g, PARAM_ROM_NAME, PARAM_ADDR_WIDTH, PARAM_DATA_WITH. The Jupyter-notebook also gives an example of how these arguments are defined.

glencoe, 2022-02-17:
Implement a code generator that creates an output like that in the template and allows us to set all the fields starting with `PARAM_`

AhmadHusein5853, 2022-03-05:
closed in 00793d2b98ba7296499ca27bf65dc3cd8e95d3bb

closed 2022-03-05 by AhmadHusein5853