es-ude/elastic-ai.creator #85: Check if simple templates can be used for code generation

glencoe, 2022-05-06:
There are still quite some vhdl components that are built from scratch by using a representation of the formal grammar of vhdl. That grammar (in `elasticai.creator.vhdl.language`) was originally started to generate the vhdl implementations for precomputed components like the Sigmoid and Tanh functions, since more than 80% of the contents for these files need to be generated.
However for many other components we can just use a template and fill in a few fields, e.g., just the field for the bitwidths and entity_name. See https://github.com/es-ude/elastic-ai.creator/pull/83/files#diff-0365f931810180507db0470eda0743afec6363dc6a4fc8f95f58b5793ac177f6 for an example.

An additional advantage of templates is that they are easier to understand for hardware designers.

The following code generators should be checked as well to see if using simple templates makes more sense for them. For each of them, check whether or not to use a template. In case we should rather use a template, create a new issue for changing that code generator.

- [ ] https://github.com/es-ude/elastic-ai.creator/blob/c9d97fa95bd113aa9acb19aaea49090663da9655/elasticai/creator/vhdl/generator/rom.py#L17
- [ ] https://github.com/es-ude/elastic-ai.creator/blob/c9d97fa95bd113aa9acb19aaea49090663da9655/elasticai/creator/vhdl/generator/precomputed_scalar_function.py#L142

SuperChange001, 2022-05-06:
For this module(https://github.com/es-ude/elastic-ai.creator/blob/b664f10cbe24adf7d36b46221e45937a44237fb5/elasticai/creator/vhdl/generator/templates/mac_async.tpl.vhd), the entity name is not necessary to be changed if the RTL part doesn't change at all. The feature we have there now is cool, exactly the same as how we import an IP Core from the Xilinx library, during that process we can give a custom name to the generated module.

SuperChange001, 2022-05-06:
I might not check all of the changes, another thing I realized is this file([elasticai/creator/vhdl/generator/templates/lstm_cell.tpl.vhd](https://github.com/es-ude/elastic-ai.creator/blob/b664f10cbe24adf7d36b46221e45937a44237fb5/elasticai/creator/vhdl/generator/templates/lstm_cell.tpl.vhd)) is not the one I am using. I am using [this](https://github.com/es-ude/elastic-ai.creator/blob/b664f10cbe24adf7d36b46221e45937a44237fb5/vhd_files/static_files/lstm_cell.vhd), it would be kind of static to the generator, means nothing should change. On the contrary, I might create different lstm_cell implementations and then ask the generator to integrate each of them into a design and compare the performance of each other.

glencoe, 2022-08-17:
short answer... yes.