es-ude/elastic-ai.creator #87: creating hex arrays for Rom module

glencoe, 2022-05-07:
While the below mentioned problem has been fixed, there are still a few open points
> The new function is clear and short, but with a bug, when the hex_list only contains one element, the length of the output array should still be 2. which means you need to append a zero to it. below is an example of quick fix
```
import math
hex_list = ["1111"]

hex_number_length = len(hex_list[0])
next_exponent_for_power_of_two = math.ceil(math.log2(len(hex_list)))
if next_exponent_for_power_of_two==0:
  next_exponent_for_power_of_two = 1
number_of_missing_zeros = 2**next_exponent_for_power_of_two - len(hex_list)
res_val=hex_list + ["".join(["0"] * hex_number_length)] * number_of_missing_zeros
print(res_val)
```

_Originally posted by @SuperChange001 in https://github.com/es-ude/elastic-ai.creator/pull/83#discussion_r866670501_

SuperChange001, 2022-05-11:
1. the ROM module is also used [here](https://github.com/es-ude/elastic-ai.creator/blob/28ca7d82e8cd0fc7040f3bc30eada4367f214ae6/elasticai/creator/vhdl/generator/generator_functions_for_one_lstm_cell.py#L16)
2. In practice, we would never need an empty room. Unless a ROM contains all values with ZEROS is the use case we need. Therefore, if the array(list) is empty, the generator should warn the user.
3. Put the hex/binary formatting process into the ROM class would be perfect.
4. In addition, since we are kind of open to the task of rewriting the `rom_generation`, so I attach a pre-designed template here.
[template_rom.vhd.txt](https://github.com/es-ude/elastic-ai.creator/files/8668238/template_rom.vhd.txt)

SuperChange001, 2022-05-12:
This issue is resolved by https://github.com/es-ude/elastic-ai.creator/pull/94

closed 2022-05-12 by SuperChange001