es-ude/elastic-ai.creator #89: switch to python 3.10

glencoe, 2022-05-07:


julianhoever, 2022-05-26:
@glencoe I tried to switch to python 3.10. But I think that currently Brevitas has some problems with torch 1.11, because in version 1.11 of torch they removed the "enable_onnx_checker" parameter of the torch.onnx.export function (in version 1.10 it was still there). See this error message of a brevitas test:

```
Traceback (most recent call last):
  File "/home/juho/ESProjects/elastic-ai.creator/elasticai/creator/systemTests/brevitas_representation/test_onnx_export.py", line 60, in test_bigger_model_save_to_onnx
    br.save_to_onnx((3, 1, 97), path="bigger_brevitas_model.onnx")
  File "/home/juho/ESProjects/elastic-ai.creator/elasticai/creator/brevitas/brevitas_representation.py", line 79, in save_to_onnx
    BrevitasONNXManager.export(
  File "/home/juho/ESProjects/elastic-ai.creator/.venv/lib/python3.10/site-packages/brevitas/export/onnx/manager.py", line 137, in export
    return cls.export_onnx(module, input_shape, export_path, input_t, **kwargs)
  File "/home/juho/ESProjects/elastic-ai.creator/.venv/lib/python3.10/site-packages/brevitas/export/onnx/manager.py", line 111, in export_onnx
    torch.onnx.export(module, input_t, export_target, **kwargs)
TypeError: export() got an unexpected keyword argument 'enable_onnx_checker'
```

And sticking to torch 1.10 with python 3.10 is, as I understand, also not possible, because torch 1.10 does not support python 3.10 at the moment.

All other integration, unit and system tests (except the export ONNX tests of brevitas) passes on python 3.10.

So what should we do? Sticking to python 3.9 or ignoring the brevitas tests until brevitas fix the support of torch 1.11?

glencoe, 2022-06-03:
for future reference: we decided to drop brevitas support in favor of the new torch versions (ie. upgrade to python 3.10)

closed 2022-06-08 by glencoe