es-ude/elastic-ai.creator #128: Implementation for Convolution

telefonjoker100, 2022-08-04:
I would like to have an implementation for a Convolution from pytorch down to vhdl.
This includes a part reading the values of the pytorch model, another one mapping the values for a vhdl-template and vhdl templates, one for the kernels shift over the input and one for the kernels operation.

julianhoever, 2023-05-19:
I guess you mean a 1d convolution? I will start working on implementing a 1d convolution.

telefonjoker100, 2023-05-26:
I think it also thought about 2d-Convolution, but 1d is a good start

julianhoever, 2023-05-28:
@telefonjoker100 Okay. Then I create a separate issue for the 2d convolution, because i would say that this is to much for a single issue.

telefonjoker100, 2023-11-14:
We have the 1d-convolution ready and should probably create a pull request.

telefonjoker100, 2024-08-13:
The implementation is already online. Now the hardware needs to be tested