es-ude/elastic-ai.creator #130: Check that tests are executed correctly

julianhoever, 2022-08-10:
[image]

can you please check whether all tests in the "tests" folder are actually executed in the ci pipeline?... Just stumbled upon this test in "test_derive_data_sets"... obviously it is not a unit test and naturally it takes forever

julianhoever, 2022-08-13:
@glencoe I have now taken a closer look at which tests are executed in the workflow and on my machine when you start them manually (python3 -m unittest discover --verbose elasticai/creator/tests). Both on my machine and in workflow, the identical 202 tests are executed. Among them is the test you had found and the test passed as well. The funny thing is that when I run the tests via the run button in PyCharm, PyCharm tells me that 240 tests have been run. But even if I test the command PyCharm executes 1:1 like this in my terminal only 202 tests are executed. Maybe this is a bug in PyCharm?

glencoe, 2022-08-17:
changes in numbers are coming from the one tool counting subtests, while the other doesn't....

Let's remove that test. It is still not terminating on Mac M1 for me. Also testing whether or not you're running out of memory is not something we should do... at least not here.

julianhoever, 2022-08-26:
The test has already been removed. I will close this issue.

closed 2022-08-26 by julianhoever