es-ude/elastic-ai.creator #143: Automatic derivation of fixed point representation

julianhoever, 2022-09-15:


SuperChange001, 2022-10-21:
In general, the code looks fine, I just have one question:
https://github.com/es-ude/elastic-ai.creator/blob/d3ba49e266b2931c1b16677dd91f17a75f091501/elasticai/creator/vhdl/custom_layers/linear.py#L42-L47
You created param_quant and input_quant, assuming you wanted to have some extensional ability later, e.g., we quantize `inputs` to 1 bit while the `weights and bias` have to stay with more bits, right?

**PS: I also notice that your implementation may actually already support different precision between weights and inputs of the linear layer. We should create an issue to track and test this and then also document this as a feature ;)**

SuperChange001, 2022-10-21:
small comment,
small comment, This function is nice and helpful; can we keep it?

https://github.com/es-ude/elastic-ai.creator/blob/b1b0b5e7697992c4c53825c739e2fb2dcc903dac/elasticai/creator/examples/fixed_point_linear_layer.py#L164-L175

julianhoever, 2022-10-22:
@SuperChange001 Oh. I didn't get notified about your comment in this issue. I will reimplement the weight visualization. I implemented this just for debugging purpose. But if you find it useful I will add this again. I will also create an issue for test and document the functionality of different quantization of inputs and weights.

julianhoever, 2022-10-22:
@SuperChange001 I reimplemented and improved the parameter visualization a little bit.

SuperChange001, 2022-11-09:
shall close this issue? or do you plan to refactor some code?

julianhoever, 2022-11-10:
I am not sure what the current status is on this issue. I don't know if it is still wanted that we should be able to automatically derive the optimal fixed point representation. Because of this, I did not close this topic.

SuperChange001, 2022-11-10:
I think we should close it and then maybe create a more concrete open issue(with a user story) for the future.

julianhoever, 2022-11-13:
Okay. Then lets close it.

closed 2022-11-13 by julianhoever