es-ude/elastic-ai.creator #155: Verify the exact rounding happens on the HDL side

SuperChange001, 2022-10-31:
This is the follow-up to @julianhoever finding, we find something very interesting about the rounding mechanism that impacts the model performance interns of training behaviour:
> During writing tests for the custom modules Linear/HardSigmoid(/ReLU) I changed the way how to convert values to FixedPoint during QAT, because in my opinion there was a mistake. What I did bevor is that I used a floor to drop the fractional part and get a integer as a result. But this leads to the problem, that for negative numbers floor will round this way: floor(-3.2) == -4.0. But this is not what we want I guess. So to really just dropping the fractional part i now do something like:  float(int(-3.2)) == -3.0. Then the fixed point calculation is correct in my opinion. But if I now do the QAT like bevor the results are slightly more unstable (see loss curve). What do you think?  Do you think this is fine?

I did the math on 2's complement, I found that on the VHDL side when we apply to round, the hardware could just round to the floor. This means -1.15 will be round to -2, while -0.625 will be round to -1. 

**I don't want to change the code now, but we should compare once we can translate your linear layer to vhdl.**

glencoe, 2023-04-26:
From my perspective model and hardware behaviour should be the same, hence if we have a hw implementation that performs rounding, then we need to round the same way in that software version. So the question is not how to handle rounding in general. 

I see two exemplary use cases for rounding:
  1. I want to round the weights to fixed point such that my hw implementation 
  2. I want to translate to a vhdl component that uses `round`
in both cases hard- and software implementation should produce the same result, but in 1. no rounding needs to happen after translation.

glencoe, 2023-10-09:
With latest introduction of simulation based verification, i think we can close the issue.
We should keep comparing simulation results and sw inference results for equality in an automated fashion.

closed 2023-10-09 by glencoe