es-ude/elastic-ai.creator #166: Auto-generate the VHDL code that connects layers

SuperChange001, 2022-11-09:
The title already tells! But I can try a better template for the network layer by passing parameters layers can be assembled.

SuperChange001, 2022-11-23:
This task could be very difficult for me. I will try and prepare materials for the auto-generation. Here is some stuff I have done:
1. The hand-written network layer with documentation: https://github.com/SuperChange001/HWLib/tree/master/source/fp_network
2. its corresponding testbench:  https://github.com/SuperChange001/HWLib/blob/master/testbench/fp_network_tb.vhd

SuperChange001, 2022-11-23:
- [ ] The next step, understand how the code generator works in the creator and try to generate the code block for instantiating components, in my case, two linear layers and two activation functions.
The generated code should look like this:
```
--------------------------------------------------------------------------------
    -- Instantiate all layers
    --------------------------------------------------------------------------------
    i_fp_linear_1d_0 : entity work.fp_linear_1d_0(rtl)
    port map(
        enable => i_fp_linear_1d_0_enable,
        clock  => i_fp_linear_1d_0_clock,
        x_addr => i_fp_linear_1d_0_x_addr,
        y_addr => i_fp_linear_1d_0_y_addr,

        x_in   => i_fp_linear_1d_0_x_in,
        y_out  => i_fp_linear_1d_0_y_out,

        done   => i_fp_linear_1d_0_done
    );

    i_fp_linear_1d_1 : entity work.fp_linear_1d_1(rtl)
    port map(
        enable => i_fp_linear_1d_1_enable,
        clock  => i_fp_linear_1d_1_clock,
        x_addr => i_fp_linear_1d_1_x_addr,
        y_addr => i_fp_linear_1d_1_y_addr,

        x_in   => i_fp_linear_1d_1_x_in,
        y_out  => i_fp_linear_1d_1_y_out,

        done   => i_fp_linear_1d_1_done
    );
    
    i_fp_hard_sigmoid_2 : entity work.fp_hard_sigmoid_2(rtl)
    port map(
        enable => i_fp_hard_sigmoid_2_enable,
        clock  => i_fp_hard_sigmoid_2_clock,
        input  => i_fp_hard_sigmoid_2_input,
        output => i_fp_hard_sigmoid_2_output
    );

    i_fp_relu_3 : entity work.fp_relu_3(rtl)
    port map(
        enable => i_fp_relu_3_enable,
        clock  => i_fp_relu_3_clock,
        input  => i_fp_relu_3_input,
        output => i_fp_relu_3_output
    );
    --------------------------------------------------------------------------------
    -- Instantiate all layers
    --------------------------------------------------------------------------------
```

glencoe, 2022-12-14:
[ ] - generate instantiation 
      ```
      (<node_id> : entity work.<module_id>
        port map(
          <signal_id_0> => <node_id>_<signal_id_0>,
          ...
          <signal_id_n> => <node_id>_<signal_id_n>
        );
        ```
        for each node in the network.

closed 2023-04-26 by glencoe