es-ude/elastic-ai.creator #170: Check GitHub Workflow deprecation warnings

julianhoever, 2022-11-13:
During the runtime of our GitHub workflow, we receive the following deprecation warnings:

[image]

I guess we need to fix them soon to make sure our workflows are working as expected.

SuperChange001, 2022-11-13:
Could you fix this when you have spare hours in the next few weeks? I will focus on end-to-end integration.

julianhoever, 2022-11-17:
@SuperChange001 Sure. I will have a look at it.

julianhoever, 2022-11-19:
Done. Only a single deprecation warning left, because the `setup-poetry` action still uses an old Node.js version. It already exists a pull requests for this GitHub Action that just needs to be approved (https://github.com/Gr1N/setup-poetry/pull/32). 

[image]

closed 2022-11-19 by julianhoever