es-ude/elastic-ai.creator #240: Implement linear layer with a batch norm layer afterwards

julianhoever, 2023-04-28:
Implement a combination of a linear layer followed by a batch norm layer that translates to a single linear layer where the batch norm operation is integrated in the linear calculation.

julianhoever, 2023-04-28:
Just as a note... Combining a linear operation with a batch norm mathematically works as follows:

For affine == False (see https://pytorch.org/docs/stable/generated/torch.nn.BatchNorm1d.html):

$lin(x) = Ax+b$
$bn(x) = \frac{x-\sigma}{\sqrt{\sigma^2+\epsilon}}$
$bn(lin(x)) = \frac{Ax+b-\sigma}{\sqrt{\sigma^2+\epsilon}} = \frac{A}{\sqrt{\sigma^2+\epsilon}}x + \frac{b-\sigma}{\sqrt{\sigma^2+\epsilon}}$
This results in $lin'(x) = A'x + b'x$ with $A' = \frac{A}{\sqrt{\sigma^2+\epsilon}}$ and $b' = \frac{b-\sigma}{\sqrt{\sigma^2+\epsilon}}$

For affine == True:

$lin(x) = Ax+b$
$bn(x) = \frac{x-\sigma}{\sqrt{\sigma^2+\epsilon}} \cdot \gamma + \beta$
$bn(lin(x)) = \frac{Ax+b-\sigma}{\sqrt{\sigma^2+\epsilon}} \cdot \gamma + \beta = \frac{\gamma A}{\sqrt{\sigma^2+\epsilon}}x + \frac{\gamma (b-\sigma)}{\sqrt{\sigma^2+\epsilon}} + \beta$
This results in $lin'(x) = A'x + b'x$ with $A' = \frac{\gamma A}{\sqrt{\sigma^2+\epsilon}}$ and $b' = \frac{\gamma (b-\sigma)}{\sqrt{\sigma^2+\epsilon}} + \beta$

SuperChange001, 2023-04-28:
If the sigma and epsilon need to be calculated on FPGA with fixed point data, this could be challenging. Never the less, the square root.

julianhoever, 2023-04-28:
No. As far as I understand it, our plan is to only collect sigma and epsilon only during training, and when translating the model to VHDL, we just calculate new weights according to my (hopefully correct) formulars. So no calculation for sigma and epsilon on the FPGA side needed.

SuperChange001, 2023-04-28:
GreatSent from my iPhone

glencoe, 2023-05-04:
formula looks correct ;)

closed 2023-05-19 by julianhoever