es-ude/elastic-ai.creator #392: FixedPointConfig does too much

chrisringhofer, 2024-11-15:
It should only specify the number of total bits and number of fractional bits. The functions it provides, should rather be implemented in a module called e.g. FixedPointArithmetic. Perhaps there is a module natively implemented in Python that provides fixed point arithmetic which we could rather use. Discussed with @julianhoever

glencoe, 2024-11-15:
Actually had a look at it, but there is no stdlib module, that does what we need.

AErbsloeh, 2025-08-16:
Maybe we have also to discuss how the arithmetic should be. I am happy a little bit unhappy that we have several version of the FixedPointConfig and the func cut_x_to_integer() avoids over and underflow. This is not what the hardware really do. There we should dicuss if we want to add this ndetection into hardware designs or we remove it in software