es-ude/elastic-ai.creator #396: Fixing BatchNorm_LinearLayer

AErbsloeh, 2024-11-26:
Transforming BatchNorm Linear from Creator into Model does not work (error in training process, values are looking random and difference between real and what we get is huge)

glencoe, 2024-11-28:
Can you go into more detail wrt. error in training process? What error? How are you finishing training to generate code if there is an error?

I think we already know that the difference is due to the batch norm parameters in full res being merged with the fixed point convolution weights.

I suggest to 

1. quantize the batchnorm as well, in a first step using the same (total, frac)-bits as the convolution. 
2. Raise a warning in case the merged result is not representable with (total_bits, frac_bits) fixed point.

Sillobear, 2024-12-03:
**Previous Method:**  
1. Train a Linear layer followed by a BatchNorm layer.  
2. Fold the BatchNorm weights into the Linear layer's weights.  
3. Generate a design for the Linear layer.  

**Issues:**  
During training, the BatchNorm layer is not quantized, which can lead to quantization errors when folding the weights later. The training process appears normal.  

To investigate further, I added a second Linear layer:  
[image]  

During evaluation, we fold the weights of the BatchNorm and the first Linear layer into the second Linear layer:  
[image]  

This produces noticeably different results, indicating that folding the weights is not a viable option for our quantized BatchNormedLinear:  
[image]

Sillobear, 2024-12-03:
### Current calculation when folding:
$$\frac{(W^TX + B) + E[X]}{\sqrt{Var[x] - \epsilon}} \ast \gamma + \beta = \gamma \ast \frac{W}{\sqrt{Var[X] - \epsilon}}X + (\gamma * \frac{B - E[X]}{\sqrt{Var[x] - \epsilon}} + \beta)$$

Problem: $Var[X], E[X], \epsilon, \gamma, \beta$ aren't quantized and therefore quantization errors arise when folding the functions.

glencoe, 2024-12-04:
> **Previous Method:**
> 
>     1. Train a Linear layer followed by a BatchNorm layer.
> 
>     2. Fold the BatchNorm weights into the Linear layer's weights.
> 
>     3. Generate a design for the Linear layer.
> 

@Sillobear : please use code fence blocks instead of images in the future, that is way easier to process and work with here on github

closed 2025-01-16 by glencoe