es-ude/elastic-ai.creator #442: Inconsistent test location

julianhoever, 2025-01-08:
In the project, tests are located next to the corresponding source file e.g.
```
elasticai/creator/.../my_module.py
elasticai/creator/.../test_my_module.py
```
and in the `tests` folder in the project root.

The `tests` folder in the project root has the following structure:
```
tests/
├─ integration_tests/
├─ system_tests/
├─ unit/
```

Questions to discuss:
- Should we keep both structures or choose one?
- Where should we put new unit tests?
- If we choose the `tests/unit` folder for new unit tests, what will happen to tests for modules starting with an underscore (e.g. `elasticai/creator/nn/fixed_point/_math_operations_test.py`)?

@glencoe What do you think?

glencoe, 2025-01-08:
Thanks for bringing that up! At some point i liked having tests next to production code to make it easy to see where stuff is tested... But as things go along i got the impression that it tends to encourage tight coupling between tests and production code -> fragile tests. That's why i'd prefer keeping tests in a separate source tree.

julianhoever, 2025-01-08:
From my point of view, it makes perfect sense to have a separate `tests' folder. It is also how I organize my own projects, but we should discuss what we want to do with the tests of "private" modules. In my opinion, we should move the tests into the `tests` folder afterwards to have a consistent project structure.

glencoe, 2025-01-10:
Another thought:

I think #456 is another good example of a case where it makes more sense to have test cases closer to the production code, because what @LeoBuron is developing there is pretty isolated and it might make sense to test a lot of details and get very whiteboxy.

In my dream world, we'd have the plugin system already established and have a structure as follows

```
elasticai/
  creator/
    base_modules/
  creator_plugins/
    gradient_quantization/
      meta.toml
      src/
      test/
test/
  unit/
    base_modules/
      conv1d_test.py
      other_base_module_test.py

julianhoever, 2025-01-20:
Ahh. It seems I forgot to move the tests inside the `elasticai.creator.nn.quantized_grads.fixed_point.test` package.

PR #483 (2025-01-20): Inconsistent test location
I moved all the tests from the project's source tree into the tests folder. To do this, I made some "private" modules public by removing the leading underscore from the module name. I did not mark this as a breaking change because these modules were private before and should not be used by a user. @glencoe What do you think about this?

Closes #442
PR #486 (2025-01-21): Inconsistent test location
Closes #442
closed 2025-01-21 by glencoe