es-ude/elastic-ai.creator #444: Redundant fixed point configurations

julianhoever, 2025-01-08:
We have three different (possible redundant) concepts to represent fixed point parameters in this project:

https://github.com/es-ude/elastic-ai.creator/blob/cb1ca7845b2e87b47acc3c3f90a37177ffe09ae0/elasticai/creator/nn/fixed_point/_two_complement_fixed_point_config.py#L30-L31

https://github.com/es-ude/elastic-ai.creator/blob/cb1ca7845b2e87b47acc3c3f90a37177ffe09ae0/elasticai/creator/nn/fixed_point/number_converter.py#L17-L18

https://github.com/es-ude/elastic-ai.creator/blob/cb1ca7845b2e87b47acc3c3f90a37177ffe09ae0/elasticai/creator/nn/quantized_grads/fixed_point/_two_complement_fixed_point_config.py#L6-L7

My feeling is that this is the symptom of not having good solution for representing and dealing with fixed point numbers. This assumption is supported by the issues #392 and #400.

In my opinion we need a simple and well-tested solution that unifies all three solution (or is easy to extend if some implementation details are different).

glencoe, 2025-01-08:
I agree. I think another reason is a missing html documentation (which i'm currently changing) and a better structure wrt. where to look for/put stuff.

julianhoever, 2025-01-08:
HTML documentation for the whole tool? Or just documentation for the fixed point config? If you agree, I would like to have a detailed look at the different fixed point configs and try to come up with a more flexible implementation?

glencoe, 2025-04-04:
Sorry, please remind me why/where this is solved.

LeoBuron, 2025-04-04:
I have no idea. I think I thought that the PR #472 was the closing this issue. But that is wrong. So we should still work on this. I personally like the way I implemented it for the quantized grads. Also the one for quantized grads are directly usable but would need to be integrated for the fixed_point implementation. I personally will not drive that as I will work on quantized grads. But if someone needs help, I will help!