es-ude/elastic-ai.creator #445: Possibly deprecated `elasticai.creator.nn.binary` package

julianhoever, 2025-01-08:
In `elasticai/creator/nn` there is a package called `binary`. The last commit that actively added a feature to this package was years ago. Is this package still in use, or do we want to offer it even if we do not currently have layers for binary quantization?

@glencoe What do you think?

glencoe, 2025-01-08:
I'll (re)introduce them, but that will happen as a plugin, so this specific package is in fact deprecated. I cannot imagine anyone used it, so i'd be fine with just removing it. What do you think about that?

julianhoever, 2025-01-08:
I would also suggest removing it, because right now it looks to a user like we support binary quantization with our tool. But we don't really support it because we don't have layers and templates for this type of quantization. If you agree, I would remove it?

glencoe, 2025-01-09:
That would be great, thanks!

glencoe, 2025-01-28:
I removed it already now, while also closing #454. However i'm far far ahead of main atm, and so is the corresponding commit.

glencoe, 2025-02-15:
solved with #527

closed 2025-02-15 by glencoe