es-ude/elastic-ai.creator #537: 🚨 Bug: 'Edit This Page' Button Leads to 404 Error

Brijeshthummar02, 2025-02-02:
#### **Description**

When visiting the website, clicking on the **"Edit This Page"** button results in a **404 File Not Found** error instead of opening the expected file for editing.

#### **Steps to Reproduce:**

1.  Go to [Website URL]
2.  Navigate to any documentation page.
3.  Click on the **"Edit This Page"** button.
4.  Observe that it leads to a **404 error** instead of opening the editable file.

#### **Expected Behavior:**

Clicking on **"Edit This Page"** should redirect to the correct file in the repository, allowing users to edit the documentation.

#### **Actual Behavior:**

The page returns a **404 Not Found** error, preventing users from contributing to the documentation.

#### **Possible Causes:**

-   Incorrect GitHub repository URL linked to the button.
-   The file might have been moved, renamed, or deleted.
-   The button logic may not be correctly handling paths.

#### **Screenshots:**

*(Attach a screenshot of the 404 error if possible.)*

#### **Environment:**

-   Website URL: [Provide link]
-   Browser: [Chrome/Firefox/etc.]
-   OS: [Windows/Mac/Linux]

#### **Suggested Fix:**

-   Verify if the file path in the **"Edit This Page"** link matches the actual repository structure.
-   Check if the base URL for editing pages is correctly configured.
-   Update the logic to dynamically generate the correct paths.

#### **Additional Context:**

Any user trying to contribute to the documentation is currently blocked due to this issue. Fixing it will improve community contributions.

💡 Note:
I can fix this issue myself. Please just point me to the exact file where the "Edit This Page" button logic is defined so I can correct the path.

📌 **Please investigate and fix this at the earliest. Thanks!** 🙌

glencoe, 2025-02-03:
Hi,

the problem here is that `index.adoc` and `contribution.adoc` are currently auto-generated from the corresponding markdown files. However, the markdown files only exist for two reasons

* pypi cannot handle asciidoc
* we want a readme and contribution at the repository root, these could be adoc files though, but the tool for building the docs expects those files at `docs/modules/root/pages/`

I want to avoid

* introducing complicated exceptions for the `Edit this Page` button
* keeping duplicate content

I think the cleanest solution is to

1. replace contribution/readme with their autogenerated adoc files (using kramdown)
2. create symlinks for both under `docs/modules/ROOT/pages`
3. autogenerate a corresponding `Readme.md` for the python package distribution.

However, as especially the last point will require a little bit more effort and since we're already creating quite some overhead to auto-generate asciidoc based api documentation, I want to talk to the rest of the team again to reevaluate the decision to use asciidoc instead of rst or markdown.

Brijeshthummar02, 2025-02-03:
@glencoe yea, wait for the team response on this one.

glencoe, 2025-02-11:
I'm closing this issue as we moved to sphinx, and afaik most edit this page buttons are working.

closed 2025-02-11 by glencoe