es-ude/elastic-ai.creator #538: Documentation format

glencoe, 2025-02-03:
Hi @es-ude/core,

some time ago me and @DavidFederl suggested to use asciidoc for our documentation instead of markdown or restructured text.

The reasons to pick it over markdown were

* there is an actual specification instead of a myriad of flavors
* tables are a lot easier to read and create than markdown tables (this might just be personal preference though)
* support for (callouts)[https://docs.asciidoctor.org/asciidoc/latest/verbatim/callouts], making it easy to annotate lines of source code examples with longer explanations
* include directives allowing us to compose content from multiple files

The reasons to pick it over restructured text were

* `rst`'s (directives)[https://www.sphinx-doc.org/en/master/usage/restructuredtext/basics.html#directives] are hard to read and i had a very hard time explaining them to others
* asciidoc's documentation is easier to digest than the reStructuredText docs which seem to be all over the place (again, might just be my personal opinion)
* in short i was sure that `rst` would have a way higher entrance barrier

As the documentation of the project is growing and in the face of issue #537, I want to reevaluate the decision to possibly turn around before it's way too late. So here are reasons against asciidoc

* `rst` and `md` are way more established, especially in the C and Python communities, e.g., linux kernel is using `rst`.
* Better tooling support for `rst` and `md`, especially for python.
* **The** rst processor Sphinx is able to include markdown as well, which would allow to combine both
* Sphinx already has the ability to extract, google doc/rst/markdown styled python docstrings and generate an api documentation. Currently we're using a very tiny tool for that which i wrote on a weekend. And while (pysciidoc)[https://github.com/glencoe/pysciidoc] works ok in most cases, it is still lacking proper support for things like doctest. **And right now i don't have any idea how to include doxygen in the picture** (there is a tool, asciidoxy, that might help but it seems more or less contradictory to what antora tries to do)
* Pretty much everyone knows at least some markdown
* Pypi expects a markdown or rst readme file and cannot handle anything else
* Many of the features i thought to be important, e.g., easy inclusion of rendered `wavedrom` diagrams (for visualizing waveforms) are realizable in rst or markdown via extensions.

one more point in favor of asciidoc

* combined with the static site generator, antora, we can generate a common documentation for the entire elastic-ai project. And antora dictates a clear architecture for the docs

If you (@es-ude/core) have any opinions here, please let me know.

----
Personal opinion: As much as i dislike markdown's tables and rst's directives, the issue #537, `rst`'s superior integration with api docs and the world apparently having agreed on markdown for docs let me think we should probably switch now, before the cost of such a switch becomes even bigger.

DavidFederl, 2025-02-05:
To my point of view we should define the markdown standard we want to use and stick with it for our purposes. In this case I think it would be beneficial to just use the [GitHub Standard](https://docs.github.com/de/get-started/writing-on-github).

If we decide on a switch, the conversion of the documentation for the elastic-ai.runtime repositories should be fairly simple and can be accomplished in a relative short time span (<4 hours).

LeoBuron, 2025-02-05:
I have no opinion on that. For me markdown works great, but I might just have not written a lot or complex documentation. 
But I am also ok with ascii doc

DavidFederl, 2025-02-05:
> Tbh. i was hoping it would be even quicker using [kramdown-asciidoc](https://github.com/asciidoctor/kramdown-asciidoc)

That was the worst-case scenario. And we should have in mind, that this would affect at least 4 repositories with multiple files each:
- elastic-ai.runtime.enV5
- elastic-ai.runtime.c
- elastic-ai.runtime.cloud
- enV5-base-project
 
> [@DavidFederl](https://github.com/DavidFederl) do you happen to have a tool at hand that you're experienced with or that you prefer for generating the static site? I used read-the-docs, but that was years ago.

 I think there are [a lot of tools/workflows](https://github.com/myles/awesome-static-generators?tab=readme-ov-file#documentation) to consider because Markdown is widely used for these purposes.
We just have to settle with one.
I would suggest [Sphinx](https://www.sphinx-doc.org/en/master/usage/markdown.html) for us, because it can convert [reStructuredText](https://www.sphinx-doc.org/en/master/usage/restructuredtext/index.html), [Markdown](https://www.sphinx-doc.org/en/master/usage/markdown.html) and Python Docstrings.

glencoe, 2025-02-05:
For the creator i was definitely going for sphinx, yes

closed 2025-02-05 by glencoe